ros-industrial/ros_industrial_issues #34: Fix project description on 'Robot Operating System' Wikipedia page

gavanderhoorn, 2015-01-16:
The Wikipedia page on the [*Robot Operating System*](http://en.wikipedia.org/wiki/Robot_Operating_System) has this to say about ROS-Industrial:

> [ROS-Industrial](http://ros.org/wiki/Industrial) is a BSD-licensed “hardware-agnostic” software
development program to create a Unified Robot Description Format (URDF) for
industrial robots.

Which is at best incomplete.

This should be fixed.

Note: the [Wikipedia:Plain and simple conflict of interest guide](http://en.wikipedia.org/wiki/Wikipedia:Plain_and_simple_conflict_of_interest_guide).

robinsonmm, 2018-07-11:
Yes the page it even references has a different description.

robinsonmm, 2018-07-11:
I have updated Wikipedia. Please review.

gavanderhoorn, 2018-07-29:
And as I expected: @robinsonmm's changes were reverted by a Wikipedia editor: [Robot_Operating_System&type=revision&diff=849848080&oldid=849834783](https://en.wikipedia.org/w/index.php?title=Robot_Operating_System&type=revision&diff=849848080&oldid=849834783). Nice that the comment included also adds:

> probable copyright violation; see https://rosindustrial.org/about/description

The [current version of page](https://en.wikipedia.org/wiki/Robot_Operating_System#ROS-Industrial) does seem like an improvement over what it was in 2015 when I opened this issue though.

If we are ok with this for now then we can close this issue as `fixed`.

gavanderhoorn, 2018-08-11:
ping @robinsonmm?

robinsonmm, 2018-08-14:
I read it, it is an improvement. We do not want to be accused of plagiarizing ourselves after all.

gavanderhoorn, 2018-08-14:
Ok, let's close this one then as resolved for now.

closed 2018-08-14 by gavanderhoorn