ros-industrial/ros_industrial_issues #36: Some problems about ros industrial_calibration

13354089, 2015-11-10:
Hi, I am following you steps on http://wiki.ros.org/industrial_extrinsic_cal to install industrial_extrinsic_cal on my ubuntu 14.04, but after I cloned industrial_calibration from https://github.com/ros-industrial/industrial_calibration and then run catkin_make in my baxter_ws, it reported some errors:

[ 60%] Building CXX object industrial_calibration/industrial_extrinsic_cal/CMakeFiles/industrial_extrinsic_cal.dir/src/camera_yaml_parser.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.8/README.Bugs> for instructions.
make[2]: *** [industrial_calibration/industrial_extrinsic_cal/CMakeFiles/industrial_extrinsic_cal.dir/src/camera_yaml_parser.cpp.o] Error 4
make[1]: *** [industrial_calibration/industrial_extrinsic_cal/CMakeFiles/industrial_extrinsic_cal.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j1 -l1" failed

Could you help me with that? It has bothered me for a few days.........TAT.

gavanderhoorn, 2015-11-10:
I think it would be better to report this at the `ros-industrial/industrial_calibration` [issue tracker](https://github.com/ros-industrial/industrial_calibration/issues).

As to your question: these errors are most of the times due to your computer not having enough memory available for compilation. Are you running on an embedded platform? Or some other machine with low physical ram, and no swap enabled? If that is the case, try to compile somewhere else and / or enable swap if possible.

jrgnicho, 2015-11-10:
I agree with @gavanderhoorn , this issue seems to be caused by insufficient memory.  This [link](https://access.redhat.com/documentation/en-US/Red_Hat_Enterprise_Linux/3/html/System_Administration_Guide/s1-swap-adding.html) describes how to add swap space when more memory is needed.

13354089, 2015-11-11:
Thank you for all you guys helping me, now the problems is solved. @jrgnicho  @gavanderhoorn

sampsonh, 2015-11-11:
hi,i meet some problems about industrial_extrinsic_tutorials,when i done the first example(Multi_camera_calibration),i following the instrument to modify the file all,but i get the wrong data,can you give me some advice about the calibration of two kinect @jrgnicho @gavanderhoorn@ Christina gomezc

gavanderhoorn, 2015-11-11:
@sampsonh: can you please not hijack threads for your own issues?

Please open a new issue on the `industrial_calibration` [issue tracker](https://github.com/ros-industrial/industrial_calibration/issues), or send an email to the [ROS-Industrial mailing list](https://groups.google.com/forum/#!forum/swri-ros-pkg-dev).

gavanderhoorn, 2015-11-11:
Seeing as you've already opened #37, I'll remove the off-topic comments here.

gavanderhoorn, 2015-11-11:
Can this be closed, @13354089?

sampsonh, 2015-11-11:
why the industrial_extrinsic_calibration-indigo-devel  pkg do not contain the  cameras_to_robot and moveit_config ,when i want to do the industrial_extrinsic_tutorial,so it confuse me ,anyone who can help me ?can you give me some advice@13354089 @jrgnicho @gavanderhoorn

closed 2015-11-15 by 13354089