ros-industrial/ros_industrial_issues #40: Installation instructions (on ROS wiki) are outdated

gavanderhoorn, 2016-04-25:
The [Industrial/Install](http://wiki.ros.org/Industrial/Install) page makes multiple references to ROS Hydro, and states it is the *latest version*.

This (these) page(s) need to be updated. Ideally they'd be rewritten to be (ROS) release independent, as far as that is possible.

gavanderhoorn, 2016-04-25:
Triggered by [ROS Jade compatible with ROS-Industrial Hydro?](http://answers.ros.org/question/232797/ros-jade-compatible-with-ros-industrial-hydro) on ROS Answers.

gavanderhoorn, 2017-02-10:
And: [ROS-Industrial support for Kinetic](http://answers.ros.org/question/254327).

zultron, 2017-11-09:
The top result for Google searches like "ROS Industrial install" and "ROS Industrial recommended release" is that wiki link pointing to Hydro, [apparently EOL][1].  It would be great to clear this up.

[1]: http://wiki.ros.org/Distributions

ipa-nhg, 2018-07-11:
Added versions tags http://wiki.ros.org/Industrial/Install, not sure if it is efficient but is the way to support several distros instructions.

Probably make more sense create some set of released packages (metapackages) to install, for example, all supported robots using:

`sudo apt-get install ros-$ROS_DISTRO-industrial-robots`