ros-industrial/ros_industrial_issues #41: Check all xacros for compatibility with in-order processing

gavanderhoorn, 2016-05-24:
From [ros/wiki/xacro - Processing Order](http://wiki.ros.org/xacro#Processing_Order):

> Classicly Xacro first loads all includes, then processes all property and macro definitions and finally instantiates macros and evaluates expressions. Thus, **later property or macro definitions will override previous ones**. Additionally, the conditional tags, `<if>` and `<unless>`, have no effect on macro or property definitions nor the inclusion of additional files.

> Since ROS Jade, Xacro provides the command-line option `--inorder`, that allows to process the whole document in read order. Hence the latest definition of a property or macro seen **so far**, will be used.

And:

> Because `--inorder` processing is much more powerful, in future version beyond Jade, the new processing style will become default and you should check the compatibility of your xacro files.

Files processed in-order and the current processing should result in identical files.

This will have to be checked for all xacros that we ship.

ipa-led, 2018-07-11:
Checked for motoman, abb, fanuc and universal_robot.

I used the linked python script to check them all.

[urdf_diff.txt](https://github.com/ros-industrial/ros_industrial_issues/files/2183819/urdf_diff.txt)

ipa-nhg, 2018-07-11:
Thanks @ipa-led
The issue can be closed! :+1:

closed 2018-07-11 by ipa-nhg