ros-industrial/ros_industrial_issues #44: See if Creative Commons licenses make more sense for support/description licenses

gavanderhoorn, 2016-11-05:
Using licenses specifically meant for source code (BSD, Apache, etc) for creative works such as robot models is not recommended. Creative Commons licenses might provide better protection / licensing terms for such artefacts.

We should investigate whether support packages should be relicensed.

gavanderhoorn, 2017-08-11:
This is not high priority, but I still feel it would be nice if we could investigate this somehow.

@ipa-mirb, @minlingc: is this something you could perhaps help with?

minlingc, 2017-08-11:
I will have the team look into it. Reading it briefly it does have some positive points for support packages...

gavanderhoorn, 2017-08-11:
One thing to keep in mind is that we often reuse / sub license mesh data from the manufacturer. I'm not sure how that factors in, but it's something to check.

shaun-edwards, 2017-08-14:
I've seen others use this. You have to be careful that absolutely no source
code is in the package.  CC licenses on source code is a red flag for
companies.   I also recommend using the most permissive CC license.

gavanderhoorn, 2017-08-14:
@shaun-edwards wrote:
> You have to be careful that absolutely no source code is in the package.

I think that will not be a problem for most (all?) of our robot support packages.

gavanderhoorn, 2018-05-27:
I'm going to close this. This doesn't seem to be a priority -- and I don't have the time to look into this myself -- although I do believe it's wrong to use a code-centric OSS license for assets.

closed 2018-05-27 by gavanderhoorn