ros-industrial/ros_industrial_issues #45: Add a wiki page on how to get started with implementing a IRS

mmoerdijk, 2016-11-08:
I had my first experience with ROS a little more than two months ago. Having read the ROS support page [1] I got the impression that the maintainers of the ROS packages don't want to be contacted directly (Section 4.1) and assumed this was the case for the ROS-I maintainers as well. 

After starting on implementing my own Industrial Robot Server and having some questions I decided to ask them on rosanswers[2]. Next to getting the answer I sought I also got the remark from @gavanderhoorn that I it would have been wise to contact the maintainers before implementing my own IRS. I see that this could have saved me some time, but I did not feel the freedom to do so because of [1]. 

Therefore I think It would help people to include a page on implementing a IRS comparable with [3] to the wiki with a request to contact the maintainers before starting on implementing an IRS.

[1] [http://wiki.ros.org/Support](http://wiki.ros.org/Support) 
[2][http://answers.ros.org/question/245525/industrial-robot-client-are-start-and-end-sequence-numbers-not-genenrated-by-default/](http://answers.ros.org/question/245525/industrial-robot-client-are-start-and-end-sequence-numbers-not-genenrated-by-default/)
[3] http://wiki.ros.org/industrial_robot_client

shaun-edwards, 2016-11-08:
@mmoerdijk, thanks for reporting.

I'm sorry that we weren't able to help you sooner and saved you some time.  The general ROS guidelines are a little stricter than the way we run things at ROS-Industrial.  It's hard for us to change core ROS wiki pages, but we do maintain our own that I think are a more inviting (see [here](http://wiki.ros.org/Industrial/#Contact_us.2BAC8-Technical_support) and [here](http://wiki.ros.org/Industrial/#Contribution)).

The page you reference [industrial_robot_client](http://wiki.ros.org/industrial_robot_client#Contact_us.2BAC8-Technical_support), does have a section that asks people to email us if they have any issues.  Are you suggesting something else?

FWIW, we also go out of our way to promote active participation in ROS-I (see [here](https://vimeo.com/187699083#t=750s)).  The problem is that for people who are new to ROS, there isn't one place the provides the type of info you are requesting, and even when it does, it doesn't apply to everyone.

mmoerdijk, 2016-11-09:
@shaun-edwards  Thanks for the reply. 

Yes I mean something different.  What I mean is that although there is a page on the IRC there is no specific wiki page on implementing an "industrial robot server".  I realize that there aren't so many people implementing one, but I still think that it would be good to have one. Now one has to look at the source code of multiple packages to determine how the robot server should act. 

Thanks for the video, will have a look at it. Actually I think that is part of my point, it makes starting with ROS a bit difficult. I think the only solution to this would be to hire someone to work full time on the documentation, are there any plan for that? ( I assume that [this position]( http://wiki.ros.org/Industrial/contribution#Technical_Writer) is not full time.)

shaun-edwards, 2016-11-11:
Now I understand.  Since you just went through the process would you be willing to write something up?  Perhaps you could create a tutorial to help people create their own industrial robot server.  Anyone can edit the wiki pages, you just need to sign up for an account.  There's even a [tutorial](http://wiki.ros.org/WritingTutorials) on how to write a tutorial.

Documentation is always something that can be improved.  It's also one of those positions for which it's hard to recruit volunteers.  We are pursuing funding that would support a paid documentation position(s), but this is a ways off.

mmoerdijk, 2016-11-11:
Let me see what i can do. Hope you find that funding soon though :)

mmoerdijk, 2016-11-11:
Oh did not mean to close it yet

gavanderhoorn, 2020-07-01:
As creating more `industrial_robot_client` based drivers is not something we want to encourage any more, I'm going to close this.

Many robot OEMs have much more performant and better supported external motion interfaces these days, and the usage of Simple Message and the IRC has decreased.

closed 2020-07-01 by gavanderhoorn