ros-industrial/ros_industrial_issues #46: Docker Organization

ipa-mdl, 2017-01-25:
I would like to run automated Docker builds for the default `industrial_ci` images (https://github.com/ros-industrial/industrial_ci/pull/116).

I can setup the Docker organization ("rosindustrial", hyphens are not allowed).
Does somebody object such an organization?

For the actual build, the Github organization needs to grant access to Docker:
https://docs.docker.com/docker-hub/github/#/github-organizations
Can somebody please try to set this up?
If this requires write access or other over-the-top access levels, we need to find another solution.

gavanderhoorn, 2017-01-25:
The page you link has this as one of the first notes:

> Note: Automated Builds currently require read and write access since Docker Hub needs to set up a GitHub service hook. We have no choice here, this is how GitHub manages permissions. We do guarantee nothing else will be touched in your account.

ipa-mdl, 2017-01-25:
> Note: Automated Builds currently require read and write access since Docker Hub needs to set up a GitHub service hook. 

I have tested it with my repositories:
Limited acess should work as well, but the hooks need to be set-up manually.

ipa-mdl, 2017-02-09:
bump: any other opinions/comments?

ipa-mdl, 2017-05-11:
Looks like the organization was created recently.

@JeremyZoss: Do you plan to set-up automated builds in https://github.com/ros-industrial/docker?

JeremyZoss, 2017-05-11:
@ipa-mdl: I actually have only limited visibility into the docker work.  @austinderic is currently working on docker support for us.  See his latest pull request: ros-industrial/docker#2.  Maybe Austin can chime in as to whether his work addresses this current issue?

ipa-mdl, 2017-05-11:
@JeremyZoss: I have already discovered the pull request, but thanks anyway!

gavanderhoorn, 2017-08-11:
Closing as `ros-industrial/docker` now exists.

closed 2017-08-11 by gavanderhoorn