ros-industrial/ros_industrial_issues #51: Update the robots.ros.org site to list all the ROS-Industrial supported robots

ipa-nhg, 2018-03-01:
There is already a ROS-Industrial tag but the only robot linked is the abb manipulator http://robots.ros.org/tags/#ros-industrial

gavanderhoorn, 2018-03-01:
This is a good idea, but we have to decide whether we actually want to do that, or whether we want to list 'ROS-Industrial' as a supported robot, and then link to `wiki.ros.org/industrial`.

The old robots page on the wiki had this fi.

Listing individual robots would be nice, but it's also a lot of work: most of our drivers are manipulator agnostic, which leads to the question which robots should be listed then on `robots.ros.org`: all models, or a generic 'brand'?

To put it into nrs: the Fanuc driver supports all Fanuc robots. Limiting it just to the variants supported by the released robot support pkgs would mean adding 19 supported robots.

gavanderhoorn, 2018-03-13:
@ipa-nhg: thoughts?

ipa-nhg, 2018-03-13:
I see your point, too much effort and difficult to maintain.  

What about a site for each robot driver(not each model)? Similar to abb http://robots.ros.org/abb-manipulators/ , where in resources we link the ROS-wiki of the driver (for fanuc: `wiki.ros.org/fanuc`) and add a link to the `wiki.ros.org/industrial` in the general description. what do you think? 

In case we need to list all the supported robot models, I suggest wait until we split the drivers into separate repos. The Readme or the ROS site of the driver should contain a list or a table of tested and supported robots. Personally I prefer have the information only once and link it.

gavanderhoorn, 2018-05-27:
My suggestion would be to find a middle ground: list all supported *brands* of industrial robots, but not the individual variants. That is probably also how people/users will be searching for 'robot support'.

That would be what we had on the old `wiki.ros.org/Robots` page (one entry per support mfg) and would not result in a lot of work. Somehow the entries we had on the `Robots` page did not survive the migration to the new site.

We had some good logos for each of the entries on the old `Robots` page, perhaps the new repository ([ros-infrastructure/robots.ros.org](https://github.com/ros-infrastructure/robots.ros.org)) still/already contains those.

Old listings included:

 - abb
 - fanuc
 - motoman/yaskawa
 - universal robots

The old page can be found here: [wiki/Robots/Backup](http://wiki.ros.org/Robots/Backup). The macros no longer work, but it's still informative. Images were added in ros-infrastructure/roswiki#136, and everything was removed in ros-infrastructure/roswiki#228.

@ipa-nhg: is this something you could take care of?

gavanderhoorn, 2018-05-27:
Btw: tbh I'm not too thrilled by the design of the `robots.ros.org` page. Lots of blacks and the fonts are way too big.

If any of the RIC managers has some resources available, perhaps spending a few hours helping upgrade / streamline the design a bit would be well-spent?

ipa-nhg, 2018-05-31:
Yes! I can do that, update the wiki.ros.org/Robots site to list the supported brands.

gavanderhoorn, 2018-06-19:
@ipa-nhg: have you had a chance to look into this?

This might be a good task for WRID (acronym I just made up for World ROS-I Day).

ipa-nhg, 2018-06-20:
unfortunately I didn't have time. :confused: 

Thanks for the proposal, I think also that this is a good task for the WRID. 
I would like also to check the online material that we have about supported hardware and status of the software and unify the documentation.

Hardware support docu:
- I also found this site http://wiki.ros.org/Industrial/supported_hardware ( the redirect link from https://rosindustrial.org/ - Developer - Supported Hardware), is this up-to-date? or should also be reviewed? 

Software status:
- we can add to each repo a README like this one: https://github.com/ipa320/cob_robots/blob/indigo_dev/README.md (at least ROS Distro support)
- or create a common README for all the ros-industrial supported repos information (I personally prefer this option)
- also we can investigate how we can filter the repositories.ros.org site to list only ROS-Industrial packages (http://repositories.ros.org/status_page/ros_indigo_default.html)

gavanderhoorn, 2018-06-20:
Could I ask you to please open a new issue for the other points you raise? Let's not turn this into a forum thread.

ipa-nhg, 2018-07-17:
Merged https://robots.ros.org/tags/#ros-industrial

closed 2018-07-17 by ipa-nhg