ros-industrial/ros_industrial_issues #52: Proposal: standardise joint and link naming across all robot support pkgs

gavanderhoorn, 2018-03-04:
The current -- implicitly -- accepted practice is to either use the `<link_N; joint_N>` naming convention for links and joints, or use something that is closer to what the manufacturer uses. An example of the first is Fanuc, while the KUKA and Motoman packages follow manufacturer naming (`joint_aN` for KUKA and `link_A` and `joint_A` with `A <- [s, l, u, r, b, t, e]` for Motoman).

The idea behind this practice is to allow for some measure of familiarity to users that have these robots and are used to the naming convention of a specific manufacturer.

However, there is a significant disadvantage to allowing this variation in names: it breaks the abstraction and vendor-independence that ROS(-Industrial) tries to create.

An example of this can be seen when using `ros_control`: in order to switch to a different robot, the list of `joint`s configured for use with a particular controller needs to be checked and updated when switching from a robot with 'standardised' naming and one without. This leads to maintenance work and potential mistakes.

A second example would be top-level xacros that compose a work cell out of a number of components, and as part of that combine a base robot macro with an EEF and several pieces of a dress-out. The former attaches to either the last `link` in the chain, `tool0` or the new `flange` frame, while the latter would connect to whatever `link` frame represents that part of the physical robot it is mounted on (ie: some `link`, not necessarily the first or last one).

---

The proposal is to standardise `link` and `joint` naming across all robot support packages for current and future xacro macros to `link_n` and `joint_n` with `n <- N` (ie: `1, 2, 3, 4, 5, ..`), regardless of naming conventions used by a manufacturer.

The main advantage is to avoid having to update `joint` and `link` names in configuration and composite/top-level xacro files and code.

A disadvantage is that for manufacturers that do use a custom naming scheme a translation procedure  or lookup table would need to be created. For most manufacturers however, this would be relatively trivial (KUKA: "drop the 'a'", Motoman: slightly more difficult, probably: "incrementally number joints and links in their natural order from base to tip").

gavanderhoorn, 2018-03-04:
Standardising these names, together with the already standardised `base`, `flange` and `tool0` frames would remove the last bit of 'implementation detail' from the kinematic structure of a particular robot variant in robot support packages.

ipa-nhg, 2018-03-13:
+1 for the standardization. link_n and joint_n sounds good to me.

is it also possible add a suffix or prefix? thinking on a system with two arms (right and left)...

gavanderhoorn, 2018-03-13:
@ipa-nhg wrote:
> is it also possible add a suffix or prefix? thinking on a system with two arms (right and left)...

yes, of course. This proposal is just about the base name for `link`s and `joint`s. Pre- and suffxes are orthogonal to that.

See also [Coordinate Frames for Industrial Manipulators](https://github.com/gavanderhoorn/rep/blob/coord_frames_ind_robots/rep-xxxx.rst) (gavanderhoorn/rep#2) for a REP about this I will submit to `ros-infrastructure/rep`. Initially that was not going to say anything about names for `link`s or `joint`s, but depending on the discussion in this issue I may add that after all.

gavanderhoorn, 2018-05-27:
This is going to be part of the REP linked in my previous comment.

Closing this, but if any further discussion is needed feel free to keep commenting.

closed 2018-05-27 by gavanderhoorn