ros-industrial/ros_industrial_issues #53: Update documentation to point to new Discourse category

gavanderhoorn, 2018-03-04:
With the migration of the ROS-Industrial mailing list / Google group to the `ROS-Industrial` ROS Discourse category, all documentation and wiki pages directing users to post messages to `swri-ros-pkg-dev` needs to be updated.

I don't know of an easy way -- other than a Google search query (possibly restricted to a certain site with `site:...`) -- to find places that need to be updated.

Use this issue to track places that have already been or need to be updated in order to avoid duplication of effort.

gavanderhoorn, 2018-03-04:
Done:

 - KUKA Troubleshooting page ([here](http://wiki.ros.org/kuka_rsi_hw_interface/Troubleshooting?action=diff&rev1=3&rev2=4))
 - Fanuc Troubleshooting page ([here](http://wiki.ros.org/fanuc_driver/Troubleshooting?action=diff&rev1=33&rev2=34))

minlingc, 2018-03-23:
more..
 - [ ]  http://wiki.ros.org/industrial_core --> point 3 contact us
 - [ ]  http://wiki.ros.org/universal_robot --> point 5 contact us
 - [ ]  http://wiki.ros.org/industrial_moveit--> point 3 contact us
 - [ ]  http://wiki.ros.org/Industrial/Tutorials --> Point 1

gavanderhoorn, 2018-03-23:
I've merged your comment with the starting post @minlingc. You can actually edit it as well.

gavanderhoorn, 2018-03-23:
I've created a wiki snippet that contains the correct text that we should include in the pages that need this, instead of editing the same text over and over.

See [industrial/contact_us_tech_support](http://wiki.ros.org/industrial/contact_us_tech_support).

I've updated the `fanuc` page with it: [diff](http://wiki.ros.org/action/diff/fanuc/indigo?action=diff).

Remove the current `Contact Us / Technical Support` section from a page and add:

```moinmoin
<<Include(industrial/contact_us_tech_support)>>
```
the snippet includes both the heading and the text, so no need to add / keep the heading.

gavanderhoorn, 2018-04-10:
@130s: thanks for updating the `industrial_core` page on the wiki. Could I only ask you to use the includable snippet? That way we keep phrasing consistent and can update it in one go if we ever need to.

:+1:

130s, 2018-04-10:
Hope now it's updated properly. Nice to have such a template!

gavanderhoorn, 2018-05-27:
Updated all pages listed in the OP.

If anything else comes up, post a new issue or comment on this one.

closed 2018-05-27 by gavanderhoorn