ros-industrial/ros_industrial_issues #54: Travis integration broken?

ipa-mdl, 2018-06-02:
At least for `industrial_ci` and `ros_canopen` the travis jobs get triggered, but Github does not show the status anymore.

gavanderhoorn, 2018-06-03:
I'd updated some security settings for ros-industrial/docker#14. Apparently Travis was making use of the fact that we previously didn't have those settings enabled.

Should be fixed now.

ipa-mdl, 2018-06-03:
I have supposed something like this..
Works again, thanks!

closed 2018-06-03 by ipa-mdl

gavanderhoorn, 2018-06-03:
Must say I'm not a fan of Travis needing so much access to 'all' of our stuff though. But that is apparently something Github needs to fix.

ipa-mdl, 2018-06-03:
With the new Github Apps integration access can be granted for each repository.
At least for user repositories.