ros-industrial/ros_industrial_issues #55: Document the status of the software (maintenance, ci builds, distro support..)

ipa-nhg, 2018-06-22:
We can create a common file as overview of the status of all the ROS-I repos, similar to this template:
 https://github.com/ipa320/cob_robots/blob/indigo_dev/README.md  

Also can be part of each repo README.md

ipa-nhg, 2018-06-22:
issues related to the https://github.com/ros-industrial/ros_industrial_issues/issues/51#issuecomment-398731310 discussion

gavanderhoorn, 2018-06-25:
Suggestion: add this as a section to existing readmes.

One file per organisation (ie: `ros-industrial`) would be nice, but could suffer from reduced visibility (which repository is going to host it?).

AustinDeric, 2018-07-06:
yes, i was going to suggest this as well!  Each README needs at least a travis build badge.

**We also need to turn on cron jobs in travis** because i restarted a few repos that i still have access to and they recently failed due to dependency issues: https://travis-ci.org/ros-industrial-consortium/godel

It would also be great to have a single page (possibly this one?) to get a quick overview as to what needs attention.  It would be very difficult to go into each repo and view the status.  I wouldn't mind taking on that task.

ipa-mdl, 2018-07-11:
IMHO all READMEs should state (for each branch!):
* which distro is supported (e.g. jade is now EOL)
* link travis jobs 
* link buildfarm jobs

A consolidated overview of all repositories would be great.

In `ros_canopen` I manage stable branches (e.g. `kinetic` and `lunar`, point to last released version) and development branches (e.g. `kinetic-devel`).

ipa-nhg, 2018-07-26:
@ipa-mdl I am agree with your comment, the main problem is that not all the repos are following this  rule for the branches. 

At least I would like to have a consensus about the template, use it (that will imply update it) or not should be a decision of the repository maintainers.  I updated my previous version for ros_canopen as template https://github.com/ipa-nhg/ros_canopen/blob/UnifyReadme/README.md , let me know your opinion and feel free to propose other format.

In parallel I will create a common Readme file for all the ROS-I supported packages as a general overview, but probably simpler ( as @AustinDeric already said, this will be very difficult if we pretend to support all repos, all branches, all distros.... )

gavanderhoorn, 2018-07-26:
My 2 cents: the currently suggested content comes across as chaotic. That is not necessarily a problem with the content, but seems to be more a result of the limitations of the various markup formats used in the readmes. Markdown just doesn't really support showing this in a nice way, unless we want to go into embedding raw html.

I'm still not entirely convinced that listing all of this information is actually beneficial (it is a duplicate of what is shown on the ROS wiki after all), one of the few 'aesthetically pleasing' ways this sort of info is visualised is the [moveit readme](https://github.com/ros-planning/moveit/blob/kinetic-devel/README.md): no button in a row: no support. Button in a row? Supported, and build status is reflected by the button. Button is immediately a link to the job. Columns show supported ROS releases.

Perhaps we can do something like that.

I would also very much recommend the proposed content be shown *after* things like the repository overview and first few sections. This may be personal, but I'd rather first understand what a repository has to offer than be shown a (potentially very large) table with stuff I might not even be interested in at first.

As to an overview of everything: something to figure out is where to host that. There are no organisation level readmes or descriptions (that allows sufficient space), so that would seem to require us to put it into a repository or potentially the ROS wiki. Alternatives could of course also be used.

ipa-nhg, 2018-07-26:
For the individual READMES, lets keep open the discussion about the format. Probably @ipa-mdl will propose a nice alternative soon....

For everything, I can imagine the following options to host the overview:

- ROS Wiki 
- The README file of a "general" repo maybe make sense in industrial_metapackages (use it for a general vision of the ROS-I software) or rename and use this ros_industrial_issues or just create a new one.
- The github Wiki of a "general repo" (again industrial_metapackages , ros_industrial_issues or new one). Although my experience with ROS wiki and tables is not good, we can try it.

About the content, we should investigate the way to have as less duplicated information as possible, I prefer always the option of having the information only once and link it. I hope I have time soon to come up with a proposal ...