ros-industrial/ros_industrial_issues #56: Migrate the package.xml files to format 2

ipa-nhg, 2018-08-09:
(or make more sense wait and migrate to format 3 directly http://www.ros.org/reps/rep-0149.html ?)

gavanderhoorn, 2018-08-11:
I think first getting everything to version 2 would be better. Changes to get to 3 are minimal, especially with the packages that we have: we don't have dependency groups, don't release into ROS2 (yet) and the ABI spec is nice, but not something of immediate concern.

gavanderhoorn, 2018-08-11:
@ipa-nhg: could you perhaps update the issue with a list of pkgs that need to be updated?

ipa-nhg, 2018-08-23:
Lists of packages to be migrated:

- industrial_core
    - [ ] industrial_msgs
    - [ ] industrial_core
    - [ ] industrial_utils
    - [ ] industrial_deprecated
    - [ ] simple_message
    - [ ] industrial_robot_client
    - [ ] industrial_trajectory_filters
- universal_robot
    - [ ] ur_description
    - [ ] ur3_moveit_config
    - [ ] ur_msgs
    - [ ] ur_bringup
    - [ ] universal_robot
    - [ ] ur_gazebo
    - [ ] ur10_moveit_config
    - [ ] ur5_moveit_config
    - [ ] ur_kinematics
    - [ ] ur_driver
- abb_experimental
    - [ ] abb_irb7600_support
    - [ ] abb_ irb4600_support
    - [ ] abb_irb1200_support
    - [ ] abb_irb52_support
    - [ ] abb_irb6650s_support
- motoman_experimental
    - [ ] motoman_robot_pkg_gen
    - [ ] motoman_experimental
    - [ ] motoman_mh5_ikfast_manipulator_plugin
    - [ ] motoman_mpl80_moveit_config
- staubli_experimental
    - [ ] staubli_tx60_gazebo
    - [ ] staubli_rx160_gazebo
    - [ ] staubli_tx90_gazebo
    - [ ] staubli_tx60_support
    - [ ] staubli_experimental
    - [ ] staubli_tx90_support
- robotiq
    - [ ] robotiq_s_model_articulated_gazebo
    - [ ] robotiq_modbus_rtu
    - [ ] robotiq_action_server
    - [ ] robotiq_s_model_visualization
    - [ ] robotiq_joint_state_publisher
    - [ ] robotiq_s_model_articulated_gazebo_plugins
    - [ ] robotiq_s_model_control
    - [ ] robotiq_c2_model_visualization
    - [ ] robotiq_s_model_articulated_msgs
    - [ ] robotiq_modbus_tcp
    - [ ] robotiq_force_torque_sensor
    - [ ] robotiq_ethercat
    - [ ] robotiq
    - [ ] robotiq_c_model_control
- staubli
    - [ ] staubli_rx160_moveit_config
    - [ ] staubli_rx160_moveit_plugins
    - [ ] staubli_rx160_support
    - [ ] staubli_resources
    - [ ] staubli
- motoman
    - [ ] motoman_gp12_support
    - [ ] motoman_gp7_support
    - [ ] motoman_gp8_support
    - [ ] motoman_mh50_support
    - [ ] motoman_mh12_support
- industrial_moveit
    - [ ] stomp_plugins
    - [ ] stomp_moveit
    - [ ] constrained_ik
    - [ ] stomp_test_kr210_moveit_config
    - [ ] industrial_moveit_benchmarking
    - [ ] industrial_collision_detection
    - [ ] stomp_test_support
    - [ ] industrial_moveit
    - [ ] stomp_core

ipa-nhg, 2018-08-23:
I create a .rosinstall file with all the ros-industrial repos which contains ros packages, cloned from the rosinstall file and commanded:
`grep -R "<package>"`

I think will be useful push somewhere the rosinstall file (the industrial_metapackages repo could be an option or just create a new one industrial_basics? industrial_helper? industrial_helper_tools? ), also we can work on some basic bash script to check things like this.

gavanderhoorn, 2019-06-29:
For anyone that wants to work on this during wrid19: please coordinate and announce the repository/packages that you're going to work on just so we can avoid duplication of effort.

dejaniraai, 2019-07-02:
I will start with the industrial_core packages

dejaniraai, 2019-07-02:
> I will start with the industrial_core packages

I have made a PR for this package. I will move this issue back to the unassigned so that someone can deal with the other packages.

fmauch, 2019-07-02:
I'll start working on universal_robot

Fern-and-dough, 2019-07-02:
I'll start working on abb_experimental

ipa-led, 2019-07-02:
@ipa-nhg robotiq packages have through re-factoring. All package.xml have been updated to version 2 for kinetic devel.

gavanderhoorn, 2019-07-02:
@ipa-led: I've updated the OP.

Fern-and-dough, 2019-07-02:
Working on Staubli as well

gavanderhoorn, 2019-07-09:
I've updated the OP to show the current state of things and unassigned the issue itself.

Thanks @wahlent, @fmauch, @dejaniraai, @Fern-and-dough and @trentwall for working on this and contributing your PRs! :+1:

ipa-hsd, 2019-10-17:
https://github.com/ros-industrial/stomp_ros already has format 2 for the `melodic-devel` branch

ipa-nhg, 2020-07-01:
seems to be completed, can this issue be closed?

gavanderhoorn, 2020-07-01:
If all referenced PRs have been merged, then yes, it would seem this can be closed.

Thanks for the reminder.

closed 2020-07-01 by ipa-nhg