ros-industrial/ros_industrial_issues #57: Add license information badget to readme files

ipa-nhg, 2018-08-09:
see https://github.com/ros-industrial/fanuc/pull/244

gavanderhoorn, 2018-08-14:
Example:

[image]

130s, 2018-12-17:
Although I've been happily using https://github.com/ros-infrastructure/rospkg/pull/143 (and its depended PR https://github.com/ros-infrastructure/rospkg/pull/133) in the internal repos for months, they aren't merged yet so need some setup. I'm happy to help if that's desired.

gavanderhoorn, 2018-12-18:
@130s: thanks for the offer.

> they aren't merged yet so need some setup

are there any really involved steps, or would it "just" be a matter of merging the PR locally?

130s, 2019-01-01:
>>     they aren't merged yet so need some setup
>
> are there any really involved steps, or would it "just" be a matter of merging the PR locally?

I'd say it depends on the platform.
- Where dependency is only minimally installed, it usually needs some adhoc setting due to https://github.com/ros-infrastructure/rospkg/issues/142. 
  - In my experience this is the case for CI where I want to use base Docker image where dependency is only sparsely pre-installed.
- If dependency is well-installed, you may use the PR as is.
  - Without even merging, you can use just by setting `PATH` to `script` and `PYTHONPATH` to `src` folder in `rospkg`.

gavanderhoorn, 2019-06-29:
This issue currently lacks an overview of the specific repositories that we'd like to add these badges too. That would first need to be added.

ipa-hsd, 2019-07-02:
https://github.com/ros-industrial/universal_robot/pull/436

gavanderhoorn, 2019-07-02:
Thanks for the many PRs @ipa-hsd.

gavanderhoorn, 2019-07-02:
I've unassigned you for now.

ipa-nhg, 2021-09-29:
@ipa-hsd @gavanderhoorn I checked all the PRs and all of them are merged or closed because they were superseded by other commits.

I will close this issue for now. @ipa-hsd thanks a lot for your contribution 👍🏽

closed 2021-09-29 by ipa-nhg

ipa-hsd, 2021-09-29:
There was some issue with https://github.com/ros-industrial/robotiq/pull/156 with multiple licenses being added. But seems like no action has been taken for it.

gavanderhoorn, 2021-09-29:
`robotiq` has been inactive for a very long time. See my additions to the README: https://github.com/ros-industrial/robotiq/commit/a8c61490285fd5732e6e71694f636e61bb58f323.

Your PR was merged though I believe.