ros-industrial/ros_industrial_issues #58: Create new repository to hold templates and helper scripts

ipa-nhg, 2018-09-06:
It would be great if we have a central (non-ros code) repository to hold templates for generic documentation, scripts to generate this documentation, rosinstall files....

Some related discussions:
- https://github.com/ros-industrial/ros_industrial_issues/issues/26#issuecomment-408705825 
- https://github.com/ros-industrial/ros_industrial_issues/issues/56#issuecomment-415399720
- https://github.com/ros-industrial/ros_industrial_issues/issues/26
- https://github.com/ros-industrial/ros_industrial_issues/issues/57
- https://github.com/ros-industrial/ros_industrial_issues/issues/39
- https://github.com/ros-industrial/ros_industrial_issues/issues/55

Some name proposals:

gavanderhoorn, 2018-09-06:
I would actually suggest to create multiple repositories, each with a name that clearly identifies what it stores.

That way we don't have to come up with a generic name ("helper") and it's obvious what goes where.

As to the specific issues linked:

 - #57 is linked because of the "license crawler", correct? We could leverage https://github.com/ros-infrastructure/rospkg/pull/143 for the actual crawling. Not sure how much more automation we want to create for adding the badges: it's rather dependent on the current content of the readme where it should all go. It would make sense perhaps though to write a short Python script that could generate the Markdown snippets.

 - #39: at first I thought it wouldn't make sense to store a copy of PR and issue templates, but seeing as we can probably use the GH API to automate configuring the various repositories, it actually would.

 - #55: this could potentially be added to [industrial_metapackages](https://github.com/ros-industrial/industrial_metapackages).

 - #56: same (`industrial_metapackages` is already about 'variants').

 - #26: we could put this in the same repository as the one for #39.

Levi-Armstrong, 2018-12-17:
@gavanderhoorn I have modifying the moveit_ci to be a little more generic to support the ros-industrial repositories which uses travis-ci directly instead of industrial-ci. Would you recommend creating a repository for this. It currently only two files [1](https://github.com/Levi-Armstrong/tesseract-1/blob/removeIndustrialCI/.travis_utils.sh) [2](https://github.com/Levi-Armstrong/tesseract-1/blob/removeIndustrialCI/.travis.sh)?

gavanderhoorn, 2018-12-17:
> @gavanderhoorn I have modifying the moveit_ci to be a little more generic to support the ros-industrial repositories which uses travis-ci directly instead of industrial-ci. Would you recommend [...]

Can you provide a little more background as to what your motivation was? Would it not have been possible to make changes to `industrial_ci` instead of creating yet another variant of `industrial_ci` (which `moveit_ci` sort-of is)?

We might want to avoid duplication of effort here, as this would be the third repository/set of scripts that would need to be maintained.

gavanderhoorn, 2019-08-26:
Somewhat later: I've created [ros-industrial/wrid_scripts](https://github.com/ros-industrial/wrid_scripts) to store the script(s) used during last WRID.

cbw36, 2020-07-07:
I'd like to work on this for WRID20, can I be assigned to it?

Also, can I get some insight into what specific documentation, scripts, and rosinstall files you are referencing so I can get started?