ros-industrial/ros_industrial_issues #59: Packages for Melodic Migration

robinsonmm, 2018-12-18:
Proposed list for Packages to be Migrated to Melodic

Name - Status
1. Industrial Core - PR
2. Industrial Calibration - Working Code
3. ros_qtc_plugin - Done
4. Basler/Astra/RealSense - not started
5. Robot Drivers (ABB, Motoman, Fanuc) - not started
6. Industrial MoveIt! - ?
7. Descartes - not started
8. industrial_training - ?
9. Universal_Robot - ?
10. Godel - not stated

robinsonmm, 2018-12-19:
Agreed. We have access to Universal, Yaskawa, ABB, and a Fanuc, though maybe not the best candidate. We also have a Kuka iiwa in house as well, and a working updated driver that should work for both Kinetic and Melodic. 

We will rely on community testing and submission of issues should problems arise for any additional or as problems arise for the variants of the makes (vintage and model).

robinsonmm, 2019-02-08:
@gavanderhoorn @Levi-Armstrong has a release pending your review I believe on ros-i core.

ipa-nhg, 2021-09-29:
@robinsonmm is this issues still relevant?

I reviewed the list of pending repositories from your old list and checked which of them have melodic support (best case a dedicated branch. And that is the result:

Industrial Core
- [ ] Melodic Branch
- [ ] Release

Industrial Calibration
- [ ] Melodic Branch
- [ ] Release

Fanuc 
- [ ] Melodic Branch
- [ ] Release

Motoman
- [ ] Melodic Branch (no but using kinetic-devel, no need)
- [ ] Release

ABB
- [ ] Melodic Branch  (no but using kinetic-devel, no need)
- [ ] Release

UR
- [ ] Melodic Branch
- [ ] Release

Industrial MoveIt!
- [ ] Melodic Branch
- [ ] Release

industrial_training
- [ ] Melodic Branch
- [ ] Release

The repos Basler/Astra/RealSense, Descartes and Godel are not part of the ros-industrial organization. 

From my point of view, the only potential pending tasks for the issues are the release for melodic of the following repositories:

- [ ] motoman 
- [ ] universal_robot
- [ ] industrial_calibration
- [ ] industrial_moveit

But not sure if this is required for all of them, I suggest to create for each of them (if still relevant) a new issue and close this meta-issue.

gavanderhoorn, 2021-09-29:
`industrial_moveit` will not see a release.

`industrial_calibration` is undergoing extensive refactoring.

`universal_robot` is already Melodic compatible.

and `motoman` as well.

I believe we can close this issue.

We can track progress for individual packages on the respective trackers.

closed 2021-09-29 by gavanderhoorn

robinsonmm, 2021-09-29:
Thanks Gijs, agreed.