ros-industrial/ros_industrial_issues #60: Should we add Probot/Stale to the ros-industrial organization?

Levi-Armstrong, 2019-02-14:
I am interested in adding this to several of the repositories that I maintain but wanted to reach out and see if it was ok to add this service to the organization. It does not force every repository to use it, only the ones that add a particular file to the repository similar to travis ci.

Levi-Armstrong, 2019-02-14:
Here is the link: https://probot.github.io/apps/stale/

gavanderhoorn, 2019-02-14:
It depends a bit on the permissions the app needs.

Personally I'm not a fan of allowing blanket access to all org data for apps, and some do ask quite a bit of access for no apparent reason.

Levi-Armstrong, 2019-02-14:
I should of looked deeper but you can add it to individual repositories. The two permissions below are the only ones it requests.

- Read access to metadata and single file
- Read and write access to issues and pull requests

Levi-Armstrong, 2019-02-14:
Is it OK if I add it to the trajopt, tesseract, and ros_qtc_plugin repository?

gavanderhoorn, 2019-02-15:
If you're comfortable with that level of access, then sure, go ahead.

Levi-Armstrong, 2019-02-15:
This is what it shows me. It not very clear.

[image]

gavanderhoorn, 2019-02-17:
yes, that looks like you add it to the organisation and then activate/configure it for specific repositories.

That would be similar to what I've seen with other "apps" (like Travis).

Not sure there is anything that can be done about that though, short of integrating them manually (which for some reason is considered inefficient and is made rather difficult).

Levi-Armstrong, 2019-02-19:
Oh, Is there any issue with the permission that have to be granted?

gavanderhoorn, 2019-02-25:
> Oh, Is there any issue with the permission that have to be granted?

No, they're probably fine.

gavanderhoorn, 2019-06-29:
Closing as it seems we've reached a conclusion.

@Levi-Armstrong: I'm not seeing any additional integrations. Did you end up installing Probot or Stale?

closed 2019-06-29 by gavanderhoorn