ros-industrial/ros_industrial_issues #61: Prefered Repository structure supporting ros and ros2?

Levi-Armstrong, 2019-03-27:
We are in the process of converting several package to support both ros and ros2. In some of the repositories a large portion of the code can easly be made to build for ros and ros2, example shown [here](https://github.com/Levi-Armstrong/tesseract-1/blob/feature/ros2/tesseract_collision/CMakeLists.txt). For other packages that leverage action, services, topics it becomes messy with a lot of #ifdef. We have been discussing internally possible options described below and we are interested to get feedback and if there is alternative solution that have not been listed.

- Option 1
  - Create three repositories.
    - One for the core packages that can be easily built with ros and ros2.
    - One containing ros packages
    - One containing ros2 packages, which should be the same package in ros just upgraded to ros2.
- Option 2
  - Create two repositories
    - One for the core packages that can be easily built with ros and ros2.
    - One for both ros and ros2 packages. In this repository there would be a branch for ros(master) and a branch for ros2(master). Then label the release branches something along the lines of ros-kinetic and ros2-crystal.

Levi-Armstrong, 2019-03-27:
@gavanderhoorn

gavanderhoorn, 2019-06-18:
Discussion took place here: [ROS2 Transition Strategy](https://discourse.ros.org/t/ros2-transition-strategy/9154).

gavanderhoorn, 2019-06-29:
I'm closing this as the discussion took place somewhere else and we're not expecting any further discussion here.

closed 2019-06-29 by gavanderhoorn