ros-industrial/ros_industrial_issues #64: WRID19 wrap-up

gavanderhoorn, 2019-07-14:
Wrap-up of WRID19:

 - [ ] review items in the [under review](https://github.com/orgs/ros-industrial/projects/3#column-5781462) column
 - [ ] determine status of work for items in the [in progress](https://github.com/orgs/ros-industrial/projects/3#column-5781461) column
 - [ ] remove `wrid19` label from items in the [unassigned](https://github.com/orgs/ros-industrial/projects/3#column-5781459) column
 - [ ] remove `wrid19` label from participating repositories
   - not before open PRs have been merged
 - [ ] close [World ROS-Industrial Day 2019 - Issue board](https://github.com/orgs/ros-industrial/projects/3)

gavanderhoorn, 2019-07-14:
@ipa-nhg @ipa-hsd @ipa-led @ipa-mdl @ipa-thz 
@Levi-Armstrong @robinsonmm @jrgnicho @JeremyZoss 
@dejaniraai 

I've got a script ready that will automatically take care of `3` (ie: remove the label from unassigned items).

I plan to run this script on Tuesday (16th) @ 10am CEST.

If you don't want items (ie: issues and PRs) in your repositories to be included in that run, let me know before that time.

ipa-mdl, 2019-07-14:
:+1:  for clean-up.

The [unassigned](https://github.com/orgs/ros-industrial/projects/3#column-5781459) contains a number of PRs that are actually waiting to be reviewed.
So the ToDo list should better start with "review items in the under unassigned column".

gavanderhoorn, 2019-07-14:
> The [unassigned](https://github.com/orgs/ros-industrial/projects/3#column-5781459) column contains a number of PRs that are actually waiting to be reviewed.
> So the ToDo list should better start with "review items in the under unassigned column".

not sure I understand.

I plan to remove the `wrid19` label from both PRs and issues that are currently in the `unassigned` column. If there are PRs that were submitted as part of WRID19, they should not be in the `unassigned` column, but in the `under review` column.

If WRID19 PRs are in `unassigned` that would be something to correct.

ipa-mdl, 2019-07-14:
> If WRID19 PRs are in unassigned that would be something to correct.

That's what I meant.
Currently, this column contains 13 PRs.
At least the ones that were opened/updated during WRID19 should be moved.

(I will check  the label assignments in industrial_ci and ros_canopen)

gavanderhoorn, 2019-07-17:
Slightly delayed, but I've ran the script to remove the `wrid19` label from items in `unassigned` (task 3).

gavanderhoorn, 2021-10-06:
Closing. 2019 was a long time ago already.

closed 2021-10-06 by gavanderhoorn