ros-industrial/ros_industrial_issues #65: Migrate to Travis.com

gavanderhoorn, 2019-10-05:
Travis has been migrating to Github Apps integration for quite some time now.

I recently received an email that I could start migrating "my" repositories to the Apps integration.

Github has slightly nicer integration with Travis through the Apps integration (availability of the *Checks* tab in PRs fi) and I'd like to migrate all repositories on `github/ros-industrial` and `github/ros-industrial-consortium`, starting with `fanuc` and `fanuc_experimental`.

I'm not aware of any downsides to this for our use-cases.

@Levi-Armstrong @ipa-nhg @dejaniraai: I'd like your OK for this and/or your thoughts.

gavanderhoorn, 2019-10-08:
If there are no comments by 2019-10-14 EoB (CEST) I'll start/proceed with the migration.

dejaniraai, 2019-10-09:
Hi @gavanderhoorn, if there are no downsides to this migration, and if the Github Apps CI tool seems to be what will be supported in the future, then I don't see any issue with starting the migration.

gavanderhoorn, 2019-10-14:
I've migrated a nr of repositories already.

@ipa-mdl: I'm about to migrate `industrial_ci` and `ros_canopen`.

Would you want me to hold off on that, or should I go ahead. It's literally no more involved than clicking a few buttons.

ipa-mdl, 2019-10-14:
Go ahead :)

gavanderhoorn, 2019-10-14:
Should be all done.

gavanderhoorn, 2019-10-14:
All repositories on `ros-industrial` that had `travis-ci.org` integration have been migrated.

ipa-mdl, 2019-10-14:
> All repositories on ros-industrial that had travis-ci.org integration have been migrated.

:+1:
Next step: update all ReadMe files to point to the new site and icons.

gavanderhoorn, 2019-10-14:
Yes, indeed.

Did you just volunteer for that ? ;)

gavanderhoorn, 2019-10-14:
> > All repositories on ros-industrial that had travis-ci.org integration have been migrated.
> 
> +1
> Next step: update all ReadMe files to point to the new site and icons.

And also done ;)

jrgnicho, 2019-10-15:
@gavanderhoorn thanks for all the work on this

ipa-nhg, 2020-07-07:
I checked the ros-industrial-consortium repositories and I found 9 with a travis configuration. For all of them I added or modified the badge of the readme. Unfortunately some of them have still to be activated to be built with the travis-ci.com system. Here the list:

- trajopt_ros (already pointing to travis-ci.com)

Active repos and the update of the readme works:
- [ ] bezier : [PR open](https://github.com/ros-industrial-consortium/bezier/pull/175)
- [ ] tesseract_ext : [PR open](https://github.com/ros-industrial-consortium/tesseract_ext/pull/16)

Not active repos, and travis-ci.com site empty:
- [ ] CAD-to-ROS: [PR open](https://github.com/ros-industrial-consortium/CAD-to-ROS/pull/123)
- [ ] descartes_tutorials: [PR open](https://github.com/ros-industrial-consortium/descartes_tutorials/pull/18)
- [ ] reuleaux: [PR open](https://github.com/ros-industrial-consortium/reuleaux/pull/62)

Require that an organization owner activate the builds:
- [ ] descartes: [PR open](https://github.com/ros-industrial-consortium/descartes/pull/246)
- [ ] godel: [PR open](https://github.com/ros-industrial-consortium/godel/pull/249)
- [ ] packml: [PR open](https://github.com/ros-industrial-consortium/packml/pull/53)

gavanderhoorn, 2020-07-07:
I've reviewed all of the PRs @ipa-nhg opened.

`tesseract_ext` is for the maintainers there to merge.

`packml` needs a comment by @shaun-edwards.

gavanderhoorn, 2021-02-24:
I'm closing this as Travis CI will be phased out.

See #74.

gavanderhoorn, 2021-02-24:
Thanks for all the work @ipa-nhg :+1:

closed 2021-02-24 by gavanderhoorn