ros-industrial/ros_industrial_issues #66: Migration to ur_robot_driver & deprecation of ur_modern_driver

gavanderhoorn, 2019-11-23:
Meta-issue for the migration to `ur_robot_driver` and the deprecation of `ur_modern_driver`.

ipa-nhg, 2020-07-01:
@gavanderhoorn what is the status of this issue? A more generic question, from your point of view what should we do with the pending issues and PRs of the deprecated repository `ur_modern_driver`?

gavanderhoorn, 2020-07-01:
I was sure I had a list of things to do and I had added that list here, but I can't find it any more.

`ur_driver` and `ur_modern_driver` have been marked as EOL and deprecated, both in their manifests as well as the wiki. The old tutorial ([this one](http://wiki.ros.org/universal_robot/Tutorials/Getting%20Started%20with%20a%20Universal%20Robot%20and%20ROS-Industrial)) has been marked as obsolete and also points to `ur_robot_driver`.

Deprecation of all drivers except `ur_robot_driver` has been announced on ROS Discourse: [Deprecation of ur_modern_driver](https://discourse.ros.org/t/deprecation-of-ur-modern-driver/11629).

`ur_driver` will be removed from `ros-industrial/universal_robot` as part of ros-industrial/universal_robot#448.

`ur_modern_driver` will be kept around for CB1 and CB2 users (`ur_robot_driver` is not compatible).

gavanderhoorn, 2020-07-01:
> A more generic question, from your point of view what should we do with the pending issues and PRs of the deprecated repository `ur_modern_driver`?

We should address them as far as resources are available and we can replicate them.

CB1 and CB2 users cannot use `ur_robot_driver`, as it requires a CB3 (RTDE is needed). So from all the drivers in `ros-industrial`, `ur_modern_driver` is the only option for anything `<` CB3.

gavanderhoorn, 2020-07-01:
As to whether this issue has been sufficiently addressed, I'm not sure.

I would probably suggest for someone to do a final check for references to `ur_driver` and/or `ur_modern_driver` which aren't accompanied by either a deprecation notice/warning or which don't make it clear those references are kept for historical purposes.

Perhaps a good candidate for WRID20.

ipa-nhg, 2020-07-01:
> As to whether this issue has been sufficiently addressed, I'm not sure.
>
> I would probably suggest for someone to do a final check for references to `ur_driver` and/or `ur_modern_driver` which aren't [...]

I volunteer myself

ipa-nhg, 2021-09-29:
@gavanderhoorn I vote for closing this issue.

For ur_modern_driver the README says clearly that it is deprecated https://github.com/ros-industrial/ur_modern_driver#deprecation-notice
And the package ur_driver was deleted for the branch melodic-devel-staging, once it is merged into melodic-devel it will be completely clear that it is deprecated.

gavanderhoorn, 2021-09-29:
Ok.

Thanks for checking.

We still get questions about this from time-to-time, but we can't pre-empt all of those I believe.

closed 2021-09-29 by gavanderhoorn