ros-industrial/ros_industrial_issues #68: Investigate use of .zae for pkgs with Collada meshes

gavanderhoorn, 2020-06-06:
As per subject.

Assimp 5 supports loading `.zae`, which is the compressed version of Collada.

Afaict, "only" zip compression is supported, but this could already save up to 90% on some meshes. As Collada is a popular mesh format for robot support packages which include more detailed meshes.

Ubuntu Focal comes with Assimp `5.0.1`, so it could be interesting to see whether this would allow us to reduce installed size of robot support packages.

traversaro, 2020-06-07:
I do not know how many ROS-Industrial robots model are expected to be used with Gazebo, but given that Gazebo use its own Collada-parser instead of Assimp, note that probably any model that uses `.zae` will not be able to be loaded in Gazebo.

gavanderhoorn, 2020-06-07:
I would almost say "that's too bad for Gazebo then, they shouldn't have used a custom parser", as that is exactly why you shouldn't do that. It essentially means that Gazebo doesn't properly support Collada `1.5`.

But of course, I can't say that.

So we'll have to take this into account.

I'm not closing this immediately as I'll see what the status of Collada support is in Gazebo.

gavanderhoorn, 2021-09-29:
We're not going to do this.

Seems there is too much variation in loaders being used, and compressed Collada is not supported by all of them.

Oh well.

closed 2021-09-29 by gavanderhoorn