ros-industrial/ros_industrial_issues #69: WRID20 candidates

gavanderhoorn, 2020-07-01:
Use this issue to suggest other topics, subjects and issues for consideration in the `wrid20` run this year.

gavanderhoorn, 2020-07-01:
@fmauch: FYI.

fmauch, 2020-07-01:
From a ur_robot_driver I would throw in the following issues:
 * joint_limits (UniversalRobots/Universal_Robots_ROS_Driver#198)
 * action server state reporting frequency (UniversalRobots/Universal_Robots_ROS_Driver#151)
 * prefix log messages (UniversalRobots/Universal_Robots_ROS_Driver#96)
 * setup_check_utility  (UniversalRobots/Universal_Robots_ROS_Driver#66)
 * Finish integration of new structure (UniversalRobots/Universal_Robots_ROS_Driver#97)
 * Update UR moveit package (ros-industrial/universal_robot#430, ros-industrial/universal_robot#431)

gavanderhoorn, 2020-07-01:
> * Update UR moveit package ([ros-industrial/universal_robot#430](https://github.com/ros-industrial/universal_robot/issues/430), [ros-industrial/universal_robot#431](https://github.com/ros-industrial/universal_robot/issues/431))

for these specifically I have a few ideas.

The MSA v2 does some strange things I would like to avoid, so this will require some manual work.

We should not use the Gazebo-support created by the MSA v2 for instance.

This will take 5 mins for someone familiar with MoveIt and how things are "ideally" done, but may take quite some back-and-forth for someone with less experience.

As fixing the MoveIt configs is one of the last things part of ros-industrial/universal_robot#448, I'm probably going to claim this for myself -- even though I might not get to it before or on WRID20.

schornakj, 2020-07-02:
Do additional repos need to be explicitly added to the WRID2020 project in order for their issues to show up on the project board?

ipa-nhg, 2020-07-02:
@schornakj yes, as linked repositories. If you don't have rights for that, list here the missed ones and we will add them.

fmauch, 2020-07-06:
As I'm not overly familiar with github features: Is there any chance we can get the https://github.com/UniversalRobots/Universal_Robots_ROS_Driver repo to the board? As far as I understood the docs it has to be done on the project board's side. Is this possible as the repo lives in another organization's namespace? Will it be enough, if I tag the respective issues with a`wrid20` label?

gavanderhoorn, 2020-07-06:
> Is there any chance we can get the https://github.com/UniversalRobots/Universal_Robots_ROS_Driver repo to the board? As far as I understood the docs it has to be done on the project board's side. Is [...]

No. That's not possible.

Only repositories in the `ros-industrial` organisation can be linked to the project board.

gavanderhoorn, 2020-07-06:
@fmauch: what we can do is add a *Note* to the board (so that's not an issue). But then the content of the node would be a reference to the issue.

I'll do that for the issues you referenced.

fmauch, 2020-07-06:
Cool, that looks promising :-)

gavanderhoorn, 2020-07-06:
@schornakj: would you want the `yak` repositories to be linked to the board?

gavanderhoorn, 2020-07-06:
@fmauch: the only downside is that changes in the status of the cards will not be reflected in the timeline of the issue (ie: you won't see automatic updates of the status (unassigned, in progress, etc). 

So that requires a bit more manual work to manage those cards.

schornakj, 2020-07-06:
> @schornakj yes, as linked repositories. If you don't have rights for that, list here the missed ones and we will add them.

> @schornakj: would you want the `yak` repositories to be linked to the board?

Sorry for the late reply. Yes, if we could link [`yak`](https://github.com/ros-industrial/yak) and[ `yak_ros`](https://github.com/ros-industrial/yak_ros) that would be excellent.

gavanderhoorn, 2020-07-06:
> Sorry for the late reply. Yes, if we could link [`yak`](https://github.com/ros-industrial/yak) and[ `yak_ros`](https://github.com/ros-industrial/yak_ros) that would be excellent.

Done.

> e: I tagged some issues that fit the WRID2020 theme, so let me know if I need to do something different to link the issues to the board.

Could you please use the formatting for the label as specced [here](https://github.com/ros-industrial/wrid_scripts/blob/7673e7f8291d3e71f01e523ca635a6c6444fd0f2/create_wrid19_labels.sh#L33) (colour, label and description)? That way the issues you label will show up in the automated search, making things much easier to manage.

mpowelson, 2020-07-06:
Are we adding [noether](https://github.com/ros-industrial/noether) or was that decided to be too obscure? There are a few relatively low hanging fruit issues on there that seem like good candidates.

gavanderhoorn, 2020-07-06:
@mpowelson wrote:
> Are we adding [noether](https://github.com/ros-industrial/noether) or was that decided to be too obscure? There are a few relatively low hanging fruit issues on there that seem like good candidates.

I'd be happy to have noether on the list.

It's up to you.

mpowelson, 2020-07-06:
I would add it. Either way I will probably work on it.

gavanderhoorn, 2020-07-06:
If you could add some cards for the issues you'd like to be part of WRID20, that would be appreciated.

schornakj, 2020-07-06:
> Could you please use the formatting for the label as specced [here](https://github.com/ros-industrial/wrid_scripts/blob/7673e7f8291d3e71f01e523ca635a6c6444fd0f2/create_wrid19_labels.sh#L33) (colour, [...]

Did it!

mpowelson, 2020-07-06:
@gavanderhoorn I don't think I can label things since I am not a maintainer on that repo. I can comment on them though.

gavanderhoorn, 2020-07-06:
Oh, that's unfortunate.

I can allow you to label issues. Would that work?

mpowelson, 2020-07-06:
@gavanderhoorn Yes it looks like I can now. Thanks.

gavanderhoorn, 2020-07-06:
Just added the new one.

Please also make sure to add those labelled issues to the board. You should be able to do that.

gavanderhoorn, 2020-08-27:
Closing as WRID20 was already 2 months ago.

closed 2020-08-27 by gavanderhoorn