ros-industrial/ros_industrial_issues #70: ROS2 version of Tolomatic servo driver

schornakj, 2020-11-05:
@ben-greenberg and I have been working on a [WIP ROS2 version](https://github.com/schornakj/tolomatic/tree/fix/ros2-rosdep-defs) of the [ROS-I Tolomatic servo driver package](https://github.com/ros-industrial/tolomatic). We want to consolidate our work in an "official" open-source repo, so I'd like to coordinate with the ROS-I organization maintainers on a good way to do this.

Right now the package is somewhat entangled with ROS-specific functionality, since it was originally written exclusively for ROS1. Ideally there should be a ROS-independent pure-C++ driver package and separate thin ROS/ROS2 nodes that expose servo control and status through services and topics, but it'll take some more work to organize things to that point.

One approach would be:
- Use the existing `tolomatic` repo for the ROS-independent libraries.
- Make a new `tolomatic_ros` repo for the ROS1 packages.
- Make a new `tolomatic_ros2` repo for the ROS2 packages.

I've been using a similar approach elsewhere (Yak/yak_ros/yak_ros2, for example) and it seems to work well.

We also need to add a new repo maintainer to the `tolomatic` repo, since I think @wpmccormick isn't active anymore.

gavanderhoorn, 2020-11-05:
I can add you as an admin to that repository if you'd like.

Does @wpmccormick no longer work at SwRI, or is he busy with other things?

I'd like to avoid just stepping in and removing people's access if it's not needed.

schornakj, 2020-11-05:
@wpmccormick no longer works on SwRI. Ben reached out to him [here ](https://github.com/ros-industrial/tolomatic/issues/7)and it seems like he's busy with other work in any case.

I think it would be OK if I were an admin for that repo.

wpmccormick, 2020-11-06:
By all means, remove me as admin. Joe has my full confidence.

BTW ... having fun doing Water/Wastewater systems. So much easier :)

gavanderhoorn, 2020-11-06:
@schornakj: I've setup two teams and made you member of both. The first team has push access, the other admin. Add ppl to the appropriate teams as you see fit. You're also team manager of both teams.

Let me know when creating those new repositories would be necessary.

schornakj, 2020-11-06:
@gavanderhoorn Thank you very much!

gavanderhoorn, 2020-12-13:
Can this be closed @schornakj ?

schornakj, 2020-12-13:
@gavanderhoorn Yes, we have the right access to make the changes we need. I'll let you know separately when we're at a point where we need a new repository.

closed 2020-12-13 by gavanderhoorn