ros-industrial/ros_industrial_issues #71: Travis CI: Builds have been temporarily disabled for private and public repositories due to a negative credit balance.

ipa-mdl, 2020-12-13:
I now get a banner

> Builds have been temporarily disabled for public repositories due to a negative credit balance. Please go to the Plan page to replenish your credit balance or alter your Consume paid credits for OSS setting.

I have now unchecked this options:

[image]

ipa-mdl, 2020-12-13:
Ah, Travis changed its billing policy: https://blog.travis-ci.com/2020-11-02-travis-ci-new-billing
And we only get 10K credits = 1000 min for all non-OSS-approved ROS-Industrial repositories.

gavanderhoorn, 2020-12-13:
Yes, this is a major problem for `ros-industrial` and `ros-industrial-consortium`.

> And we only get 10K credits = 1000 min for all non-OSS-approved ROS-Industrial repositories.

I've contacted Travis to ask for an "allotment of OSS credits", but the response was that they've stopped that for now (meaning: no OSS credits for us, or anyone afaict until they resume that).

We've already migrated some repositories to GH Actions. With `industrial_ci` that's essentially a relatively easy switch.

ipa-mdl, 2020-12-13:
I have removed all industrial_ci daily cron jobs for now.

gavanderhoorn, 2021-02-24:
We'll phase out use of Travis CI across all affected repositories.

See #74.

Closing this issue as a result.

closed 2021-02-24 by gavanderhoorn