ros-industrial/ros_industrial_issues #76: Create user story template for gathering industrial requirements

ipa-cmh, 2021-09-29:
- [ ] Create a repository for user stories
- [ ] Research available methodologies
- [ ] Create a template for user stories
- [ ] Add the first user stories from local storage

ipa-cmh, 2021-09-29:
@gavanderhoorn @robinsonmm
I have created a short preview of a possible template.
https://github.com/ipa-cmh/ros_i_user_stories

It is based on agile user stories, but I added a more dedicated description of the user and his industrial problem/use case, so we know where the story comes from.

What do you think about it?

gavanderhoorn, 2021-09-29:
Looks OK.

High level comment: any time there is the option to list multiple items (such as for *User type*), I'd suggest using an actual list, or use comma separation. Or change markup such that *robotics end-user* and the other options are clearly distinguishable from the "and". It's not easy right now to scan the file.

Did you also want to create something to capture missing features / functionality / problems?

ipa-cmh, 2021-09-29:
> High level comment: any time there is the option to list multiple items (such as for _User type_), I'd suggest using an actual list, or use comma separation. Or change markup such that _robotics [...]

Good point on machine processing.

> Did you also want to create something to capture missing features / functionality / problems?

I suppose we could have a list of necessary technologies/features/functionalities for the user stories.

gavanderhoorn, 2021-09-30:
We could consider using a YAML based format for this. Example: [robust-rosin/robust/kobuki/03660af/03660af.bug](https://github.com/robust-rosin/robust/blob/833a891c98b69beb0df0a9411f23990f02175b32/kobuki/03660af/03660af.bug).

It wouldn't need to be this extensively structured or complex, but YAML is still human readable, allows free-form text (in certain special fields) and would make this immediately machine readable.